dekuNukem/exixe_clock #2: Slot 4 not working

BenRoe, 2019-07-21:
Hello,
i soldered the exixe_clock pcb and flashed the simple firmware. 5 Slots are working, but the 4th not.
If i switch the exixe12 modules, slots 4 still not working.
I checked all solder pads. Any idea what could be the problem, or how i can debug it?
Btw many thanks for this great open source project.
[image]
[image]

dekuNukem, 2019-07-21:
Does the backlight work on that slot? If the backlight works but the tube doesn't glow, it might be the high voltage traces. And if the backlight doesn't work at all, it could mean the module itself isn't working, which could be the 3.3V power, or the incorrect SPI signal, so check those.

I have also designed an updated clock board with USB-C connectors and various other small refinements, which you can check out in the repo. The functionality is exactly the same.

BenRoe, 2019-07-21:
The backlight on slot 4 glows white and the other slots orange.
I guess it's a SPI problem?! All other slot's are working so it't the CS trace?
It's definitely not the exixe12 module. If i put it in another slot it works.
Soldering on port 25/PB12 looks good.

I would try the new pcb version, but i have to order min. 5 and already have 4 left from the previous.
If someone is interested i can give away 4x v1 pcb's.

dekuNukem, 2019-07-21:
If the backlight on slot 4 is dim white, it means the module is getting power but no SPI data. Check the continuity of SCK and MOSI signals, look for shorts and breaks, and make sure CS signal is properly soldered along the way. A logic analyzer would be very helpful, oscilloscope would be great too.

Since the new version has the same functionality as the old one, you don't actually _need_ to make another one, but of course it's up to you. Here's the [change log](https://github.com/dekuNukem/exixe_clock/blob/master/pcb/v2_changelog.txt)

dekuNukem, 2019-07-21:
Make sure PB12/Pin25 pad actually has electrical connection to the metal pin on the microcontroller, sometimes it looks like its soldered but not actually touching.

BenRoe, 2019-07-21:
Many thanks for your friendly and quick support.  
I moved the solder iron over the lower pin of the STM32 and now it works. Under the magnifier everything looked good, but i am sure it was the CS solder point on the CS pin (25/PB12).

I don't need the other 4x pcb's. Should i open another issue for the give away? If someone is interested he can comment. If you don't like this idea, please let me know.

dekuNukem, 2019-07-21:
Glad that you fixed it! This kind of invisible soldering imperfection does happen sometimes.

pcb giveaways isn't technically an "issue", but i don't really mind. So feel free if you want.

closed 2019-07-21 by dekuNukem