dekuNukem/exixe_clock #4: Time jumping around

azvero, 2019-09-20:
First of all, thanks for the great project.

On several occasions now, I've observed the clock behaving erratically, jumping around by a few seconds for a short while.  It seems to happen sporadically:  I've not noticed any pattern or yet been able to capture on video to demonstrate.

I was wondering if you might have any suggestions for how best to troubleshoot this?

My build follows the BOM exactly, including all the optional parts except for SD card holder.  Additionally, it has a Raspberry Pi 3 B+ hanging off the GPS header (TP and TX pins, no RX).  The power is supplied by the Raspberry Pi directly to the +5V rail.  The Raspberry Pi is plugged into an Anker charger capable of delivering 4A per port.  The GPS module is an M8Q.

azvero, 2019-09-20:
Oh, and it's a V1 build.

dekuNukem, 2019-09-22:
Hi azvero, the microcontroller periodically updates its internal RTC with GPS time, and if the RTC time and GPS time doesn't match, the time will jump around a bit. The frequency is defined here:

https://github.com/dekuNukem/exixe_clock/blob/ef3eac2cc35f0b0e01800e94d85808dc1e6a31e7/firmware/Inc/helpers.h#L20

Right now it's every 5 seconds, which was way too frequent. Change it to every 5 minutes or so would be much better.