dekuNukem/exixe_clock #5: Request for submitting Gerber files

jaromczyk, 2019-10-21:
Hi,
thanks for sharing this project, it is really awesome!

I was trying to import the project onto KiCad to generate the Gerber files, but I get some errors - it would be best if one was simply to download the gerber files and then getting them manufactured.

Thanks again!

dekuNukem, 2019-10-25:
I've added the gerber file in the repo, you can find it here:

https://github.com/dekuNukem/exixe_clock/blob/master/pcb/clock_v2_gerber.zip

jaromczyk, 2019-10-25:
Thank you very much! Could you do the same for the exixe modules?
Thanks again!

dekuNukem, 2019-10-25:
https://github.com/dekuNukem/exixe/blob/master/pcb/exixe12/exixe12_gerber.zip

https://github.com/dekuNukem/exixe/blob/master/pcb/exixe14/exixe14_gerber.zip

jaromczyk, 2019-10-25:
Thanks!
Closing this issue.

closed 2019-10-25 by jaromczyk